LSTS/dune #42: Plan.Engine: Segfault when sending invalid plan

krisklau, 2015-11-20:
Hi,

When one sends a plan containing a transition to a maneuver ID that does not exists, a seg-fault occurs. See e.g. the following snippet to construct a dummy plan:

    // Create the maneuver
    IMC::Goto goto_man;
    goto_man.lat = 10.0;
    goto_man.lon = 60.0;
    goto_man.z   = 10;
    goto_man.z_units = IMC::Z_ALTITUDE;
    goto_man.speed = 10;
    goto_man.speed_units = IMC::SUNITS_METERS_PS;

    IMC::PlanManeuver* pman1 = new IMC::PlanManeuver();
    IMC::InlineMessage<IMC::Maneuver> pman1_inline;
    pman1_inline.set(goto_man);
    pman1->maneuver_id = "1";
    pman1->data = pman1_inline;
    
    // Prepare maneuvers
    IMC::MessageList<IMC::PlanManeuver> manlist;
    manlist.push_back(pman1);
    
    
    
    // Set up transitions
    IMC::PlanTransition* transition = new IMC::PlanTransition;
    transition->source_man = "1";
    transition->dest_man   = "2";
    
    // Prepare transitions
    IMC::MessageList<IMC::PlanTransition> translist;
    translist.push_back(transition);
   
    // Prepare the Plan Specification
    IMC::PlanSpecification plan;
    plan.description = "A net recovery test plan";
    plan.plan_id = "NetRecoveryTest";
    plan.start_man_id = "1";
    plan.maneuvers = manlist;
    plan.transitions = translist;
    
    IMC::InlineMessage<IMC::Message> inline_planSpecification;
    inline_planSpecification.set(plan);
    
    // Create planControl
    IMC::PlanControl planCtrl;
    planCtrl.op = IMC::PlanControl::PC_START;
    planCtrl.type = IMC::PlanControl::PC_REQUEST;
    planCtrl.plan_id = "NetRecoveryTest";
    planCtrl.request_id = 10;
    
    
    // Apply plan specification
    planCtrl.arg = inline_planSpecification;
    
    dispatch(planCtrl);

My backtrace:

    #0  0x00007ffff6f9f16d in std::string::compare(char const*) const () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
    #1  0x00000000007c7d6a in std::operator==<char, std::char_traits<char>, std::allocator<char> > (__lhs=..., 
        __rhs=0xcf9c63 "_done_") at /usr/include/c++/4.8/bits/basic_string.h:2521
    #2  0x000000000091613c in Plan::Engine::Plan::sequenceNodes (this=0x7fffcc000bc0)
        at /home/krisklau/uavlab/dune/src/Plan/Engine/Plan.cpp:518
    #3  0x0000000000915c75 in Plan::Engine::Plan::secondaryParse (this=0x7fffcc000bc0, cinfo=..., ps=..., 
        imu_enabled=false, state=0x11dfcb0) at /home/krisklau/uavlab/dune/src/Plan/Engine/Plan.cpp:454
    #4  0x0000000000914b1e in Plan::Engine::Plan::parse (this=0x7fffcc000bc0, supported_maneuvers=0x11dfc40, 
        cinfo=..., ps=..., imu_enabled=false, state=0x11dfcb0)
        at /home/krisklau/uavlab/dune/src/Plan/Engine/Plan.cpp:127
    #5  0x000000000090e6da in Plan::Engine::Task::parsePlan (this=0x11df810, plan_startup=true, ps=...)
        at /home/krisklau/uavlab/dune/src/Plan/Engine/Task.cpp:747
    #6  0x000000000090e035 in Plan::Engine::Task::loadPlan (this=0x11df810, plan_id=..., arg=0x7fffb0001c80, 
        plan_startup=true) at /home/krisklau/uavlab/dune/src/Plan/Engine/Task.cpp:667
    #7  0x000000000090ef84 in Plan::Engine::Task::startPlan (this=0x11df810, plan_id=..., spec=0x7fffb0001c80, 
        flags=0) at /home/krisklau/uavlab/dune/src/Plan/Engine/Task.cpp:873
    #8  0x000000000090dd4c in Plan::Engine::Task::processRequest (this=0x11df810, pc=0x7fffb0001c20)
        at /home/krisklau/uavlab/dune/src/Plan/Engine/Task.cpp:624
    #9  0x000000000090db47 in Plan::Engine::Task::consume (this=0x11df810, pc=0x7fffb0001c20)
        at /home/krisklau/uavlab/dune/src/Plan/Engine/Task.cpp:598
    #10 0x0000000000914409 in DUNE::Tasks::Consumer<Plan::Engine::Task, DUNE::IMC::PlanControl>::consume (
        this=0x11e2eb0, msg=0x7fffb0001c20) at /home/krisklau/uavlab/dune/src/DUNE/Tasks/Consumer.hpp:53
    #11 0x0000000000898568 in DUNE::Tasks::Recipient::runCallBacks (this=0x11e0c60)
        at /home/krisklau/uavlab/dune/src/DUNE/Tasks/Recipient.cpp:109
    #12 0x0000000000898476 in DUNE::Tasks::Recipient::waitForMessages (this=0x11e0c60, timeout=1)
        at /home/krisklau/uavlab/dune/src/DUNE/Tasks/Recipient.cpp:88
    ---Type <return> to continue, or q <return> to quit---
    #13 0x00000000007e44df in DUNE::Tasks::Task::waitForMessages (this=0x11df810, timeout=1)
        at /home/krisklau/uavlab/dune/src/DUNE/Tasks/Task.hpp:487
    #14 0x000000000090fd31 in Plan::Engine::Task::onMain (this=0x11df810)
        at /home/krisklau/uavlab/dune/src/Plan/Engine/Task.cpp:1142
    #15 0x000000000089ec5f in DUNE::Tasks::Task::run (this=0x11df810)
        at /home/krisklau/uavlab/dune/src/DUNE/Tasks/Task.cpp:368
    #16 0x00000000007ec0fa in dune_concurrency_thread_entry_point (data=0x11df810)
        at /home/krisklau/uavlab/dune/src/DUNE/Concurrency/Thread.cpp:88
    #17 0x00007ffff79bc182 in start_thread () from /lib/x86_64-linux-gnu/libpthread.so.0
    #18 0x00007ffff6a0447d in clone () from /lib/x86_64-linux-gnu/libc.so.6

Kristian

rasmartins, 2015-11-20:
Please check branch bug/github-42 for a fix.

krisklau, 2015-11-20:
Thanks, confirmed to work :)

rasmartins, 2015-11-20:
Merged.

closed 2015-11-20 by rasmartins