LSTS/dune #51: Run DUNE on GLUED

aljimgar, 2016-02-17:
Hi, I have a Raspberry Pi 2 with GLUED "lauv-aux-rpi-testbed", ¿the correct form to run DUNE is copying the folder inside the sd and run the commands from the serial terminal?

josebraga, 2016-02-17:
Hi,

I assume you used ./mkread to flash the rootfs into the SD.
Usually DUNE can be started using "services dune start/stop/restart".

After initial flash, there's no DUNE content. You can copy the DUNE package (make package) to /opt/lsts, uncompress, and rename to have /opt/lsts/dune.

After that DUNE should start by default on boot, you can use "services dune start/stop/restart" or even, for debugging, you can start dune by hand /opt/lsts/dune/bin/dune -c <config file> -p <profile>

Let me know if this works as I may be forgetting something,

José

aljimgar, 2016-02-17:
Seems to be all OK, I have copied the DUNE package in the "data0" partition of the sd, inside lsts/dune. When the RPi runs, appears "Starting DUNE upgrade... done" ¿Now how can I execute tasks inside the RPi?
Thank you for your fast response.

josebraga, 2016-02-18:
Whenever a DUNE instance runs, it starts a task manager that collects the
list of tasks from a configuration file (.INI)

Then, all parameters for each task are stored, and individual threads per
task are started.

Example:

./dune -c lauv-noptilus-1 -p Hardware

Uses configuration in etc/lauv-noptilus-1.ini
That single file may include others with 'Require'.
Inside brackets you see the namespace of a task (e.g: Control.AUV.Attitude
related to the task you find following that folder tree under 'src')

By default, I believe that glued configuration probably tries to start
configuration with name 'lauv-aux-rpi-testbed' (which you can check in
variable $cfg_hostname in file /etc/config, or by running 'hostname')

That configuration is non existent so I think it won't start anything. You
can create a configuration file named lauv-aux-rpi-testbed.ini and add the
list of tasks you which to run

Do you have some application/need in mind?

aljimgar, 2016-02-18:
Ok, I will try yo create the .ini file.
I am working in the team of Javier Busquets, we are from the UPV (Valencia), we are doing an underwater glider with your platform.

Thank you!

aljimgar, 2016-02-22:
Now I have "lauv-xplore-2-aux" GLUED version with DUNE inside data0/lsts. I have created the ini file from the configuration of "lauv-explore-2", and nothing appears in the serial monitor. ¿What I am doing wrong?

josebraga, 2016-02-22:
I am a little confused with "serial monitor".

Do you have any sensor connected? What were you expecting to see in the
serial monitor ?

are you able to open a shell in the raspberry pi 2 once glued is installed?
by default (glued config) the IP is 10.0.10.183. Are you able to ping it
and ssh to it ?

If you wish to schedule a skype meeting to clean up these issues let me know

closed 2016-03-15 by aljimgar