LSTS/dune #88: Is the project dead?

rmsc, 2017-05-02:
There are countless open pull requests and no one seems to be paying attention. Are there any active maintainers around?

paulosousadias, 2017-05-02:
Not dead !
Maria is responsible now.

mariacosta, 2017-05-02:
Still active, but we are just a few active developers and with accumulated
jobs. It's true we haven't been able to get to the PRs as we should, but
it's hard to get the time to do it.

Any particular one (pull request) you need merging asap?

krisklau, 2017-05-13:
I think all of @rmsc's PRs are in need of merging, and most of them are pretty small. I can help to test some of them. 

Some of the PRs from our end (especially the FBWB-addon for ardupilot) are actually obsolete at this point, so I'll close it. 

I would also like to point out that accepting (or not accepting) PRs is critical, as new developers quickly see it as stagnation and it can reduce the development pace and adoptions to new projects. 

Maybe a more formal system where external developers, like me, can help with reviews?

mariacosta, 2017-05-13:
Hi krisklau,

Regarding the NTNU's PR's, a few months ago I addressed Fortuna about this
and he told me he would take care of those and warn me if something was
needed (testing or so) in this end. And as I recall some of them were still
marked as WIP. Maybe I got it wrong.

Nevertheless I might take you up on your suggestion. Right now we can use
the extra hands :)

This week I'll be out in work, so just give me a few days so I can take a
better look at the opened PR's and disscuss this with our team. I'll give
you an update on this as soon as I can.

Best regards,
Maria Costa

krisklau, 2017-05-13:
Yes you are right about some of our PRs are hanging on us here at NTNU to finish up :) I closed #59. The one left now is #60, which is a bit weird (and marked as WIP as you say). #55 will be closed, and that leaves #58 from me which I would like to see merged. 

Happy to help in any way I can!

closed 2017-08-17 by mariacosta