LSTS/dune #96: Using IMC in own C++ Project

TobiasTheuerkauff, 2017-08-24:
Hello, 
i want to use imc in an own c++ Project to get and send Messages from/to dune. How is that possible. I can only find a imclib for Java. Is there an imc- library for using in c++ and where can i find that?

Thanks for helping me.
Best regards
Tobias

rasmartins, 2017-08-25:
Hi,

You can just compile DUNE and use it as a C++ library to
serialize/deserialize IMC messages. If you compile DUNE using make package
the resulting zip our tarball will contain all the necessary includes and
static libraries to do what you want. Take a look at this project that does
exactly that:

https://github.com/oceanscan/ros-imc-broker

Kind regards,
Ricardo

TobiasTheuerkauff, 2017-09-23:
Hello,

thank you for your very fast reply @rasmartins! I have tried to do what you've said.

Unfortunality i couldn't resolve my problem yet...
So, first i'll give you some informations about my environment. On the one hand i work with Linux. For running Dune i use the programming environment CLion from JetBrains. Compiling and running dune works well. On the other Hand i have to use Dune in the Windows10 operating system. I have tried to use VisualStudio2017 for running dune, but there i did not get it to run. So i using CLion in Windows too. [By the Way, i think using CLion is the easiest way to run dune with Windows. There I only have to import the project from the git repository, CLion recognized automatically that it is a cmake project and will make the rest for you. Maybe that could be interesting for the dune wiki for using dune in windows?!]

But back to my problem: 
I have to get the dune-project as windows-library for using the imc-functions cause i want to develop an application for testing AUV/USV-control-algorithms. For that i need to use software, which is only available in windows.
So, now i can run dune in windows. But how can i compile dune in the way you said in your post? I can't find any solution for compiling the dune-project using make package for getting an zip-File with the needed files.

Kings regards
Tobi

TobiasTheuerkauff, 2017-10-04:
Hello,

i found the solution for my problem. i'am a newbe in this aera, so i'am sorry about to find a solution over this forum, instead of find a solution by myself!

But for all other newbes's, i'll give you my way: 
in Windows and in linuy it will be the same with the CMake-Tool:
In CMake : Configure the downloaded folder; click advanced Options - search for "Shared" - Check it- Configure it again - generate it -> you will get an lin named "libdune-core.dll" [libdune-core.so]i n your build folder! You Can find the .h-Files in the src/dune -Folder

Thank You!
Tobi

closed 2017-10-04 by TobiasTheuerkauff