LSTS/dune #97: CPU times overflows ?

gmircea, 2017-08-25:
I'm thinking of the vars in Resources::getProcessorUsage.
What happens if the tmp var that stores the /proc/stat values overflow ?  I guess you could get some large values on a long running system.
When doing the diff between new_CPU and old_CPU you'll get funny results...
Or even when computing CPU total in Core::cpuTotal().
Or am I missing something ?

rasmartins, 2017-08-25:
Hi,

If I remember correctly the tmp variable is a 64-bit unsigned integer that
holds jiffies. Since the jiffies are incremented at a fixed rate if we
consider a scheduler frequency of 1000 Hz the jiffies count will overflow
in about 50 million years of continuous operation.

Kind regards,
Ricardo

gmircea, 2017-08-28:
Ok, that sounds good. 
As an additional question, how does the kernel handles this internally, the numbers reported in _/stat/proc_ ? Are those also on 64-bits? Does it matter if on a 32bit platform or 64bit platform ?

closed 2018-03-01 by josebraga