LSTS/dune #114: C++11 as standard

krisklau, 2018-02-01:
Hi,

I was wondering if there has been any discussion about using C++11 as default for DUNE? We have been using it internally for a while, and several of the features are really useful when working with DUNE. 

Kristian

rasmartins, 2018-02-01:
It's definitely something that I've been ruminating in the last few months. From the developer point of view it's a no-brainer. However, since we (OceanScan) have a lot of systems in the field with C++98 only compilers it would be a major effort for us to upgrade those systems in order for them to run new DUNE versions. With that being said, I'm inclined to take the leap to C++14 one of these days.

krisklau, 2018-02-08:
That makes sense, and thank you for the reply. 

In the meantime, would be it an option to somehow enable this on a task-specific basis? Or the other way around, disable tasks if C++14 is not available for the compiler? This would make it possible to use the legacy compilers for most of the tasks. 

I'l close this for now as I mainly wanted your input, keep us posted.

krisklau, 2018-08-29:
Reopened to hear the status of this.

zepinto, 2019-03-03:
Now using CPP14 (master).

closed 2019-03-03 by zepinto