LSTS/dune #120: Math/Matrix: Possible error in quaternion <-> Euler angles conversions

mortenfyhn, 2018-03-07:
What is the Euler angle convention used in `Matrix::toQuaternion` and `Matrix::toEulerAngles`. The math in `toQuaternion` seems to match that for the ZYX convention, but I can't get it to work out the other way. I would suggest declaring the Euler angles convention used in the docstrings in the header, for clarity, and updating the methods accordingly.

For instance noticed that
```c++
double data[] = {1, 2, 3, 4};
Matrix q1(data, 4, 1);
q1 /= q1.norm_2();  // Normalize
Matrix q2 = q1.toEulerAngles().toQuaternion();
```
gives different `q1` and `q2` for pretty much all values of `q1`.

josebraga, 2018-03-07:
Hi,

That code was never used during my time in the lab (afaik). Since you are taking this effort to have a clean Quaternion class implementation I would (imo) delete all quaternion-related content from Matrix class (maybe euler angles as well) and keep separate, distinct classes for all of them.

Then, have a generic class to translate between those objects..

So, I believe you can delete them, fix them, or, proceed with your developments ignoring those Matrix class methods;

mortenfyhn, 2018-03-08:
I agree completely. I'll close this and continue in #119.

closed 2018-03-08 by mortenfyhn

josebraga, 2018-03-08:
I modified the quaternion to euler angles portion of the code.
The notation being used is ZYX. I have not verified other other conversion such as DCM.. when I have the time I'll write a proper test. Of course, if you conclude your classes meanwhile, I'll vote to delete these conversions in Matrix class.