LSTS/dune #128: Bus error: 10

grcasanova, 2018-05-15:
Hello,
when trying to execute a UAV simulation, upon connecting to the HTTP host, I encounter the following error:
`Bus error: 10 (core dumped)`
What could be the reason for a Bus failure and set. fault?
Thanks

josebraga, 2018-05-18:
Hi,

I have no idea what may be causing it, so can you please provide more information?

* are you running master hash https://github.com/LSTS/dune/commit/213b32f4742467ed7d56743848c991bd0b41503f ?
* which configuration?
* which OS
* was that a one-off, or does it happen persistently ?

can you describe the steps?

Thanks

grcasanova, 2018-05-19:
I am on Mac high Sierra, whenever I try running the simulator and then connecting via browser/curl the bus error comes in. I tried with other configurations, the error seem to be there
`./dune -c ../etc/x8-02 -p Simulation
`

josebraga, 2018-05-21:
I do not have a mac available and I am do not get that error in my linux.

Any chance you have the time to run it using **valgrind** or similar tool?

Thanks

grcasanova, 2018-05-29:
I can try with **lldb** on Mac and see if there is a seg fault or similar

zepinto, 2018-07-12:
Any updates on this @grcasanova ?

grcasanova, 2018-07-21:
Here is the output of lldb

```
Process 1090 stopped
* thread #2, stop reason = EXC_BAD_ACCESS (code=2, address=0x700000722000)
    frame #0: 0x00007fff5b8e604f libsystem_platform.dylib`_platform_memmove$VARIANT$Haswell + 303
libsystem_platform.dylib`_platform_memmove$VARIANT$Haswell:
->  0x7fff5b8e604f <+303>: vmovaps %ymm1, (%rdi)
    0x7fff5b8e6053 <+307>: vmovaps %ymm2, 0x20(%rdi)
    0x7fff5b8e6058 <+312>: addq   $0x40, %rdi
    0x7fff5b8e605c <+316>: vmovups (%rsi), %ymm1
Target 0: (dune) stopped.
```

adallolio, 2018-08-06:
Hi,
you can go for a Docker solution, easy and plug&play:
https://github.com/adallolio/LSTS-toolchain-macOS
This will allow to run the toolchain from Mac in few steps, you may just want to modify the Dockerfile with the clone of the repository you actually need (by default is the public LSTS repository).

Cheers
Alberto

grcasanova, 2018-08-15:
I believe using a Docker workaround is just detrimental to the project development, since this error is surely a symptom that shouldn't be neglected; if it were to happen on a real model, it could cause a crash.

Apparently it's a bad pointer allocation (raises a BAD_ACCESS exception), I will try to debug further.
I also suggest to reach out ArduPilot developers, it would be nice to add DUNE to the suggested companion computer setups.