LSTS/dune #140: MSG_SEND macro conflict

grcasanova, 2018-08-15:
Apparently the _MSG_SEND_ enum is already defined in the socket.h header, simply changing the name solves the error.

```
../dune/src/Transports/Radio/Telemetry.hpp:67:36: error: 
      expected unqualified-id
        m_tx_mesg.state = MsgState::MSG_SEND;
                                    ^
/usr/include/sys/socket.h:564:18: note: expanded from macro 'MSG_SEND'
#define MSG_SEND        0x1000          /* Send the packet in so_temp */
```

tsmarques, 2018-08-16:
Hi,

which branch are you checking, that gives you this error?

grcasanova, 2018-08-17:
Hi,
Mac OS High Sierra, I just cloned the master branch and used the system default compiler.

tsmarques, 2018-10-30:
Hi,

I was not able to get a mac to test with that version. Could you please test https://github.com/LSTS/dune/tree/issue/140

Thanks

tsmarques, 2018-11-26:
Hi,

I'll close this issue by the end of the month and assume it is fixed. If otherwise please open a new one mentioning this.

Thank you,
Tiago

closed 2018-12-07 by tsmarques