LSTS/dune #141: Algorithms/Base64: Windows compilation

oysstu, 2018-08-21:
The regex.h dependency in Algorithms/Base64.cpp is not available on Windows. Regex was added to the standard library in C++11, and can be considered as a replacement when C++11/14 is put into use.

Not a high priority issue, but having Windows compatibility is good in general for student activities and DUNE wrappers for equipment running Windows.

- Øystein

krisklau, 2018-10-23:
Ah. So when we do the switch to Cmake 3.1 and set C++ standard to 14, this problem basically goes away?

oysstu, 2018-10-23:
Well, the implementation in Base64.cpp must then be converted to the syntax in the standard library. Posix-style regex grammar is supported in the standard, so the regex shouldn't have to be changed. Haven't used regex through the standard library yet though, so I cannot comment on how easy it is to use

oysstu, 2018-10-23:
From what I can tell it should simply be something along the lines of 
```C++
    std::basic_regex<char> re("^([A-Za-z0-9+/]{4})*([A-Za-z0-9+/]{4}|[A-Za-z0-9+/]{3}=|[A-Za-z0-9+/]{2}==)$", regex_constants::extended);
    if(std::regex_match("dGVzdA==", re)){
        // Do stuff
    }
```

krisklau, 2018-10-23:
Yeah. Seems this commit (https://github.com/LSTS/dune/commit/417861589912619f1f06f5f00d9ff0d55d448ddf) introduced the library.

pmfg, 2019-02-20:
Possible fix in branch https://github.com/LSTS/dune/tree/issue/regex (need more test).

krisklau, 2019-02-21:
Isn't this just as easy a fix? https://github.com/LSTS/dune/pull/153

We just need #114 to get merged!

zepinto, 2019-02-22:
We are still testing CPP14 internally. By mid next week #153 / #156 will probably be merged.

zepinto, 2019-03-03:
Fixed in master.

closed 2019-03-03 by zepinto