LSTS/dune #173: Ardupilot-task: Bug in down velocity

krisgry, 2019-03-28:
The minus sign in the following line should not be there:
https://github.com/LSTS/dune/blob/aa489e42720bea794faea9fdc2aedcae8e550944/src/Control/UAV/Ardupilot/Task.cpp#L1831

According to mavlink, this is already given in down, as required by the EstimatedState message
https://github.com/LSTS/dune/blob/aa489e42720bea794faea9fdc2aedcae8e550944/vendor/libraries/mavlink/common/mavlink_msg_global_position_int.h#L7-L17

This bug also affects the body-frame velocities, as they are computed from the NED ones.

jffortuna, 2019-03-28:
Yep, that part of the code predates the ardupilot change to NED:
https://groups.google.com/forum/#!topic/drones-discuss/b4TutgTVoIc

krisklau, 2019-04-04:
Woah. Relevant code in ardupilot: https://github.com/ArduPilot/ardupilot/blob/master/libraries/GCS_MAVLink/GCS_Common.cpp#L3970

mariacosta, 2019-06-13:
Thanks for catching this. Already merged PR#180.

closed 2019-06-13 by mariacosta