LSTS/dune #214: Windows compilation

zepinto, 2020-05-21:
Currently, DUNE fails to compile in windows for trying to use 'unsigned long long' types:

```libSimulators.StreamVelocity.a(ModelDataStreamGenerator.cpp.obj):ModelDataStreamGenerator.cpp:(.text+0x62b): undefined reference to `std::vector<unsigned long long, std::allocator<unsigned long long> > DUNE::Parsers::HDF5Reader::getAttribute<unsigned long long>(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) const'
collect2: error: ld returned 1 exit status```.

MinGW cross-compilation toolchain (for testing):
 * https://gist.github.com/zepinto/dbbe84fa4ca6e7f59954adabc5bc3dcc

mcpca, 2020-05-21:
Just pushed a fix to bugfix/hdf5-win, I'm able to build on mingw and x64_64 Linux.
Could you confirm that it works for you?

zepinto, 2020-05-21:
Yes, this fixes the problem. Will merge to master.

commit cbf1f2cc (2020-05-21): DUNE/Parsers/HDF5Reader: Add missing template specialization.

Fixes #214.

Missing specialization for 'unsigned long long' caused linking errors
when compiling for x86_64-w64-mingw32.
closed 2020-05-21 by zepinto