LSTS/dune #267: Set log path from parameter?

krisgry, 2025-02-13:
We would like to set a custom path for the dune logs, to keep them on a different partition (e.g. a removable drive). Currently m_ctx.dir_log is hardcoded to be in ../log from the path of the dune executable.

One solution is to define a parameter (e.g. "Log directory"). Then the Daemon would have to get the parameter, and set m_ctx.dir_log (to something different than the default) if the parameter has a value. See 
https://github.com/LSTS/dune/blob/4e3d6a0f9a4d5992e2bcc6528ff8b8ebcd67ea84/src/DUNE/Tasks/Context.cpp#L52

I originally thought that we could make a `[General]` parameter, but it is not possible to have different set of General parameters for Hardware and Simulation (AFAIK?), which would be nice. 

So then I thought that the parameter could be created in `[Transports.Logging]`, and have config like this
```
[Transports.Logging/Hardware]
Log directory = /some/path 

[Transports.Logging/Simulation]
Log directory = /another/path
```
but realized that both `[Transports.Logging/Hardware]` and `[Transports.Logging/Simulation]` would have to be hardcoded into Context.cpp since config.get relies on the full name of the config section (`config.get("Transports.Logging/Simulation",.....`).  This is clearly not ideal, and most current config only include `[Transports.Logging]`.

Do you see any better ways?

zepinto, 2025-02-13:
Would it be feasible for you to use a symbolic link to the other partition
or, alternatively, mount the other partition in the [DUNE]/log folder?
We used to do this when logging to removable storage.

Kind regards,
José

krisgry, 2025-02-14:
Hi,
That should be possible. We would have to modify `maybe_upgrade` (a version of [this](https://github.com/LSTS/glued/blob/3ac9c086c77978262c0dc69aeafc916ffe3a3c27/rules/filesystem/fs/etc/rc.d/dune)) to not delete the log-symlink when upgrading, but that should be fine.

Thanks!

closed 2025-02-14 by krisgry