LSTS/dune #278: Unsupported IMU in DUNE

camila442-0, 2025-08-08:
Hello guys,
I have a problem connecting the sensors of a real vehicle. I have an IMU sensor of type MPU5060, and it doesn’t have a driver recognized by DUNE. Do you have any suggestions or solutions for this issue?

krisgry, 2025-08-12:
Implement a driver? 😃

Bernardo-Gabriel, 2025-08-20:
Hello @camila442-0 . Did you mean MPU6050? DUNE doesn't currently have a driver for that sensor. However, recently, I attempted to create a driver for the MPU9250 sensor (similar to the MPU6050), based on an Arduino library for that same sensor. You can find it in this branch: [feature/MPU9250](https://github.com/LSTS/dune/tree/feature/mpu9250/) - [src/Sensors/MPU9250](https://github.com/LSTS/dune/tree/feature/mpu9250/src/Sensors/MPU9250) - 3844a24. In that case, I tried to provide support for both I2C and SPI; using the SPI, I wasn't successful in accessing the MPU9250 magnetometer (AK8963) data (the MPU6050 doesn't have a magnetometer, so I believe you wouldn't have this problem). Notice that, in the main branch of DUNE, there is no support for SPI, yet. If you need the SPI, you may need to cherry-pick the following commits: (c669bbd, 86514af, c88d283).
Hope this helps. If you manage to implement the driver for the MPU6050 sensor, feel free to open a pull request with that. 😃